F1p/Mitsubishi-CN105-Protocol-Decode #3: HA 2025.1.x Template Warning

F1p, 2025-01-12:
> ValueError: Template error: bool got invalid input 'unknown' when rendering template '{%set mode=value_json.SystemOperationMode|lower%}{%set h_prhbt=not(states('sensor.ecodan_ashp_zone_2_heating_prohibit')|bool)%}{%set c_prhbt=not(states('sensor.ecodan_ashp_zone_2_cooling_prohibit')|bool)%}{%set wrkg=states('sensor.ecodan_ashp_zone_2_working')|bool%}{%if mode in ['defrosting','frost protect']%}defrosting{%elif (h_prhbt or c_prhbt) and not(wrkg)%}idle{%else%}{{mode}}{%endif%}' but no default was specified rendering template for entity 'climate.zone2_flow_climate_4', template: '{%set mode=value_json.SystemOperationMode|lower%}{%set h_prhbt=not(states('sensor.ecodan_ashp_zone_2_heating_prohibit')|bool)%}{%set c_prhbt=not(states('sensor.ecodan_ashp_zone_2_cooling_prohibit')|bool)%}{%set wrkg=states('sensor.ecodan_ashp_zone_2_working')|bool%}{%if mode in ['defrosting','frost protect']%}defrosting{%elif (h_prhbt or c_prhbt) and not(wrkg)%}idle{%else%}{{mode}}{%endif%}' and payload: {"HeaterFlow":16.5,"HeaterReturn":15,"FlowReturnDeltaT":1.5,"OutsideTemp":4,"Defrost":"Off","InputPower":0,"HeaterPower":0,"EstInputPower":0,"EstHeatOutputPower":0,"EstCoolOutputPower":0,"Instant_CoP":0,"Compressor":0,"SystemPower":"On","SystemOperationMode":"Off","HolidayMode":0,"FlowRate":0,"RunHours":80,"HB_ID":30}

Template warning appears in Home Assistant 2025.1.x

F1p, 2025-01-16:
Default may not be required - 6.1.2 adds dip switches, possibility now to use these to publish parameters only available to the unit

closed 2025-02-13 by F1p

thecoop, 2025-07-21:
Unfortunately I'm still getting this error on firmware version 6.2.4-h4, HA 2025.6.1

thecoop, 2025-07-21:
Ah, I've figured it out - I've disabled the zone 2 entities, as my system doesn't have a zone 2 - hence the error in the template