F1p/Mitsubishi-CN105-Protocol-Decode #15: Support long-term statistics for sensors

thecoop, 2025-07-16:
None of the sensors published by the bridge seem to have long-term statistics registered in HASS. This means that I can't use the `statistics-graph` card to display them, which is much faster to load than using the `history-graph` card (as statistics averages over every 5 minutes, rather than displaying every single value).

F1p, 2025-07-16:
It's certainly possible to add to the sensors:
```javascript
"state_class": "measurement"
```

This can be done automatically from the bridge or manually pushed in something like MQTTExplorer 

I'm not sure this is an issue, but i think it's certainly a good suggestion
I can see benefit in some values more than others - is there any downsides to making all sensors like this?

F1p, 2025-07-16:
Added in [7d26ecb](https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/commit/7d26ecb14f3bc6bf69b44a216db0dd5965703fdb) for proposed 6.3.0 release

Testing available in the [/build](https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/tree/master/build) folders

F1p, 2025-07-19:
Closing as implemented in [v6.3.0](https://github.com/F1p/Mitsubishi-CN105-Protocol-Decode/releases/tag/v6.3.0)
Thanks for raising this @thecoop

closed 2025-07-19 by F1p