F1p/Mitsubishi-CN105-Protocol-Decode #18: FTC version flapping

willholley, 2025-07-23:
I noticed today that the FTC version appears to be flapping between FTC6 and FTC2B:

[image]

I don't know whether this has any functional impact but I noticed it when trying to diagnose a problem with DHW boost not sticking.

F1p, 2025-07-23:
On boot it will ask the FTC for it's version - When it reports FTC2B it is the default (and invalid) answer because no response has yet been returned. The FTC5 is quite quick to answer, but the FTC6 is slower.
Then when it has a valid FTC version it will retain it and not ask again. 

By it changing regularly like this it suggests your device is restarting, 
The most likely reason for restart is that WiFi signal was lost for 300 seconds, it performs regular attempts to connect/re-connect before this - but if it can't find anything for 300s it will restart in an attempt to clean connect or enter the first time Access Point mode again

If WiFi was dropped, it's likely why your automations may not reach the device either

willholley, 2025-07-23:
Interesting - thanks, that's useful to know (and curious that WiFi has started playing up...)

willholley, 2025-07-23:
The uptime reported by the bridge seems solid. Is there anything that would indicate restarts / poor WiFi signal?

willholley, 2025-07-23:
It seems like DHW fast boost works but standard DHW boost is rejected. I just updated to 6.3.0 but also updated Home Assistant today so could be that one of these things has caused a problem.

The bridge reports good WiFi signal though I can see regular flapping online/offline/unavailable (since before today) in the Bridge Status sensor.

willholley, 2025-07-24:
I've swapped to a different WiFi network and the flapping has stopped for now.

closed 2025-07-24 by willholley