F1p/Mitsubishi-CN105-Protocol-Decode #20: Gen 1 dead

Callumhome, 2025-08-31:
Gen one showing orange light no access to web portal or anything tried the reset holding the button down till I get ashp ap bridge rejoin to wifi nothing working still not able to controll anything need hellpppp

F1p, 2025-08-31:
Orange is MQTT unable to connect 
Is it running in home assistant?

Callumhome, 2025-08-31:
Found the mqtt details are missing in webpage thats finally reloaded but no option for mqtt detail

F1p, 2025-08-31:
On gen1 you need to refresh the webpage a few times before it appears

F1p, 2025-08-31:
The details would of been wiped when you press & held to factory reset

Callumhome, 2025-08-31:
Press and held for rest managed to get them but it won't accept them light remains orange and unable to controll anything in Home Assistant

F1p, 2025-08-31:
Can you connect with telnet (client like PuTTY) and check the logs output?

Callumhome, 2025-08-31:
They all been rebooted all operational just this keeps going dead

Callumhome, 2025-08-31:
Goes blue green orange then completely blank 
Then back to blue

Callumhome, 2025-08-31:
Possibly in my it room.

Callumhome, 2025-08-31:
I have it connected to mqtt via home assistsnt can now see it as a device but cannot controll anything on it
All is unavailable

F1p, 2025-08-31:
All is unavailable is no reply from the heat pump
Maybe the Ecodan also needs a restart

Callumhome, 2025-08-31:
I've also done that few times now rebooted via main switch and lights still orange only got online time fo 30 seconds so its rebooted again

Callumhome, 2025-08-31:
Still doing it seems to constantly reboot i can see online time in the esp ip keeps getting to 2 mins then rebooting

Callumhome, 2025-08-31:
[image]

F1p, 2025-08-31:
It sounds like the issue reported, downgrade to 6.2.x?

Ethernet is a good option if you can, more stable than WiFi - I also swapped to that version

F1p, 2025-08-31:
Email me to arrange a swap

Philip.thomson at f1p.uk

F1p, 2025-09-03:
For Gen 1 users, this will be fixed forward - but to recover immediately you will need to downgrade to 6.2.x and then perform a factory reset (press&hold internal button) to clear the memory.

The issue is related to memory allocation on the smaller esp8266 chip